strongback/strongback-java #4: Creating a build system

WawerOS, 2015-09-05:
We need a build system to create artifact's. We should use ant as all FRC Java team's have to have it. The build.xml  should generate a zip file of the following specifications
- strongback.jar (used by robot code and deployed to robot)
- strongback-testing.jar (used by robot unit tests, not deployed to robot) 
- strongback-sources.jar
- strongback-testing-sources.jar
- JavaDoc
- LICENSE.txt, README.md, and CONTRIBUTORS files
- example Eclipse-ready projects that a user could import, build, and possibly deploy

rhauch, 2015-09-05:
Our initial [installation instructions](https://github.com/strongback/strongback-java/wiki/Installation) proposes the following structure for the directories and files in the ZIP file:

```strongback
   /java
       /ant
           build.properties
           build.xml
       /javadoc
           ...
       /lib
           strongback.jar
           strongback-sources.jar
       /lib-proc
           strongback-postprocess.jar
           strongback-postprocess-sources.jar
       /lib-tests
           strongback-testing.jar
           strongback-testing-sources.jar
           junit.jar
           junit-sources.jar
           metrics-core.jar
           metrics-core-sources.jar
       README.md
       CONTRIBUTORS.txt
       COPYRIGHT.txt
       LICENSE.txt
```

WawerOS, 2015-09-05:
Ok, so a question, where is the post-process code in the project?

rhauch, 2015-09-05:
It's not in there. It will be a separate project in the repository. It hasn't been ported from https://github.com/frc-4931/2015-Robot/tree/master/LogConverter

rhauch, 2015-09-08:
Oh, see issues #6 and #7 to add executable scripts for data post-processing and minimal help.

rhauch, 2015-09-13:
@WawerOS, are you working on this? If so, once you accept the invitation to join the Strongback organization on GitHub, we can assign you.

WawerOS, 2015-09-14:
Yep I am.

Zabot, 2015-09-15:
Its just semantics, but we should distribute `strongback-log-processor.jar` as an executable included with strongback, not as part of the library itself, since it's impossible to extend or use any other way then just executing it.

    /strongback
        /bin
            strongback.sh
            strongback.bat
            strongback-log-processor.jar
        /java
            /lib
                strongback.jar
                strongback-sources.jar
            ...

rhauch, 2015-09-15:
@Zabot, if the JAR is an executable, then yes we definitely can put it in the `bin` directory. However, if we do have a `strongback` executable (see #6) then users really would have no reason to execute the JAR directly. And if that's the case, then I'd still argue the separate `strongback-processing.jar` (or `strongback-log-processing.jar` if we prefer) should be kept out of the `bin` directory so that `strongback process ...` is the one way to post-process the log files.

rhauch, 2015-09-25:
@WawerOS, have you been able to make progress on this? This is blocking us from releasing, so it'd be great if we could get something soon.

WawerOS, 2015-09-29:
I'm sorry to ask can someone take this over school is revving up. I've pushed my changes to my repository if anyone wants to work off of that. The classpath has been giving me trouble.

rhauch, 2015-09-29:
@WawerOS can you provide a URL to your repository and branch?

WawerOS, 2015-09-29:
Right sorry, [https://github.com/WawerOS/strongback-java]()

PR #17 (2015-10-16): Create Ant-based build system
The build system can clean, compile, run tests, build JAR files for each project, and it can also build the JavaDoc and a ZIP distribution file.

Quite a few other files were touched in this PR, mostly to correct JavaDoc errors. A few other changes were made that simplified the build system or maintained consistency with names and patterns.

Closes #4
closed 2015-10-16 by rhauch