strongback/strongback-java #5: Automated builds?

AGausmann, 2015-09-06:
I am setting up a Jenkins CI server for my personal projects. If we add Strongback as a project, I think the automated builds and tests could help.

rhauch, 2015-09-08:
+1 for automated CI -- it is a must have.

But IMO we don't want to rely upon the resources of any individual. Instead, I think we should use https://travis-ci.org, which is cloud-based, free for open source communities, far easier to configure than Jenkins, and works really well with GitHub. For example, it will annotate our GitHub repository with flags that state whether it builds successfully, and we can even add hooks so that any PRs are automatically built after the PR is created *before* someone has to look at it. And it already knows about Ant, and [by default](http://docs.travis-ci.com/user/languages/java/#sts=Projects%20Using%20Ant) it will run `ant test` (tho we can [change that](http://docs.travis-ci.com/user/languages/java/#sts=Projects%20Using%20Ant) if we need to).

The only reason I didn't register our repository already was that we need a build script, so as soon as issue #4 is resolved we can add Strongback to Travis to get automatic CI.

AGausmann, 2015-09-08:
Good idea. I don't have experience in Travis, but it seems it can do everything I was planning on doing. I say we should set it up once we have a build system.

PR #22 (2015-10-19): Added support for automated continuous integration using Travis-CI.org
Set up https://travis-ci.org/profile/strongback to automatically build the PRs and attach the result of the build on the PR page.

Closes #5
closed 2015-10-19 by rhauch