strongback/strongback-java #6: Add 'strongback' executable

rhauch, 2015-09-08:
At this time there is only one thing that might correspond to a Strongback "executable": running the data post-processor (see issue #6). But we could easily plan ahead and make room for more.

We should create a "strongback" executable script that can do a number of different things. We'd need a bourne shell script for Linux and OS X and a separate but equivalent Windows batch file. These could be kept in a `script` top-level project/folder in our repository and installed into the ZIP file at `strongback/bin`. As long as users added that `strongback/bin` folder to their path, they'd just have to type `strongback ...` on the command line to use it.

Generally this would take the form:

    strongback <command> <command-param-1> <command-param-2> ...

Initially, the two commands might be `process` and `help`, so you'd also be able to run the data post-processor with:

    strongback process <path-to-data-file> <optional-path-to-output-file>

and get minimal help with:

    strongback help

Zabot, 2015-09-14:
I'll tackle this one, I think I still have some shell scripts lying around from the last season.

rhauch, 2015-09-14:
Another idea @Zabot had yesterday was to automatically add strongback to an Eclipse project created by the WPILib plugin. 

    strongback add-project <path-to-eclipse-project-dir>

See #10 for details.

rhauch, 2015-09-25:
@Zabot, have you been able to make progress on this? This is blocking us from releasing, so it'd be great if we could get something soon.

Zabot, 2015-09-25:
@rhauch Cleanly parsing the command line is difficult, especially trying to do it with shell scripts. Have a look at Zabot/strongback-java@e8961b1c08ea4c83ee8b85077cfef17bfccc8e15. I refactored `LogDecoder` to use the Apache Commons CLI Library and mushed some things about. 
I'm still trying to decide if any of that made it easier to read, but I'm thinking the best way to do the rest of the executables is using Java and Commons CLI in the same style. A separate file for each executable (new-project, add-project, etc.), each with its own `main()`, and then a single `Strongback.java` with its own `main()` that can delegate to them. That gives us the flexibility to distribute everything as either one jar, or as separate independent jars.
Doing things in Java now may also make it easier to migrate to an eclipse plugin, or not, I've never looked at it.

Zabot, 2015-09-25:
Distribution directory structure with single jar

    Strongback
        libs
            strongback-exec.jar
        bin
            strongback.sh
            strongback.bat
        java
            libs
                strongback.jar
                strongback-testing.jar
            ...

Distribution directory structure with independent jars

    Strongback
        libs
            new-project.jar
            convert-project.jar
            log-decoder.jar
        bin
            new-project.sh
            new-project.bat
            convert-project.sh
            convert-project.bat
            log-decoder.sh
            log-decoder.bat
        java
            libs
                strongback.jar
                strongback-testing.jar
            ...

Zabot, 2015-09-25:
Alternatively we can drop command line parameters and do it with a shell/bat script that expects an exact syntax. Its a little more difficult for the user, but substantially easier to maintain.

rhauch, 2015-09-25:
We should use command line parameters, but I'd rather avoid bringing a 3rd party library just to do this. For bash scripts, we should use [`getopts`](http://mywiki.wooledge.org/BashFAQ/035#getopts), which works really well and is pretty straightforward. (Note we should *not* use `getopt` because it doesn't handle args with whitespace.)

For Windows, we can pretty easily [parse them](http://stackoverflow.com/questions/14286457/using-parameters-in-batch-files-at-dos-command-line), though we may have to add extra logic if we want the usage to be similar to bash.

For Java, I have a utility class that does this, and it has a number of convenience methods to convert the string values to a few primitive types. I've not used it on strongback yet, but feel free to use it:

```
/**
 * Utility for parsing and accessing the command line options and parameters.
 * 
 * @author Randall Hauch
 */
@Immutable
public class CommandLineOptions {

    /**
     * Parse the array of arguments passed to a Java {@code main} method and create a {@link CommandLineOptions} instance.
     * 
     * @param args the {@code main} method's parameters; may not be null
     * @return the representation of the command line options and parameters; never null
     */
    public static CommandLineOptions parse(String[] args) {
        Map<String, String> options = new HashMap<>();
        List<String> params = new ArrayList<>();
        List<String> optionNames = new ArrayList<>();
        String optionName = null;
        for (String value : args) {
            value = value.trim();
            if (value.startsWith("-")) {
                optionName = value;
                options.put(optionName, "true");
                optionNames.add(optionName);
            } else {
                if (optionName != null)
                    options.put(optionName, value);
                else
                    params.add(value);
            }
        }
        return new CommandLineOptions(options, params, optionNames);
    }

    private final Map<String, String> options;
    private final List<String> params;
    private final List<String> orderedOptionNames;

    private CommandLineOptions(Map<String, String> options, List<String> params, List<String> orderedOptionNames) {
        this.options = options;
        this.params = params;
        this.orderedOptionNames = orderedOptionNames;
    }

    /**
     * Determine if the option with the given name was used on the command line.
     * <p>
     * The supplied option name is trimmed before matching against the command line arguments. Note that any special characters
     * such as prefixes (e.g., '{@code -}' or '{@code --}') must be included in the name.
     * 
     * @param name the name for the option (e.g., "-v" or "--verbose")
     * @return true if the exact option was used, or false otherwise or if the supplied option name is null
     * @see #hasOption(String, String)
     */
    public boolean hasOption(String name) {
        return hasOption(name, null);
    }

    /**
     * Determine if the option with one of the given names was used on the command line.
     * <p>
     * The supplied option name and alternative names are trimmed before matching against the command line arguments. Note that
[... 256 lines cut ...]

                config = Configuration.load(pathToConfigFile, classLoader, this::printVerbose);
                if (config.isEmpty()) {
                    print("Unable to read configuration file at '" + pathToConfigFile + "': file not found");
                    return status.complete(ReturnCode.UNABLE_TO_READ_CONFIGURATION);
                }

                printVerbose("Found configuration at " + pathToConfigFile + ":");
                printVerbose(config);

                // Adjust the properties by setting any system properties to the configuration ...
                printVerbose("Applying system properties to configuration");
                config = config.withSystemProperties(DEFAULT_SYSTEM_PROPERTY_NAME_PREFIX);
                printVerbose(config);
```

The constant and enum are defined as:

```
    private static final String DEFAULT_SYSTEM_PROPERTY_NAME_PREFIX = "STRONGBACK_";

    protected static enum ReturnCode {
        SUCCESS, UNABLE_TO_READ_CONFIGURATION, CONFIGURATION_ERROR, ERROR_DURING_EXECUTION,
    }
```

PR #15 (2015-10-05): Strongback Executable - New project and log converter
Here it is, there still is a bit of cleanup to do, but its pretty much ready to go, and it does affect #4, so I wanted to get it pushed.

* `NewProject` functions as both a `new project` and a `convert project` if the override flag is set. It doesn't set any eclipse settings (classpath variables or user libraries), that will be handled by either another command, or instructions to the user, in case their wpilib is in a strange place.

* `LogDecoder` got copied, the old one still works, but it can probably be deprecated if not removed entire [... cut]
closed 2015-10-05 by rhauch