strongback/strongback-java #9: Should non-sensor component interfaces (e.g., Motor, Solenoid, etc.) all extend Requirable?

rhauch, 2015-09-11:
Only a few of the hardware component interfaces have _writable_ (or _changable_) methods: 

* `Motor` and its sub-interfaces `LimitedMotor` and `TalonSRX`; and 
* `Solenoid` and its sub-interface `SolenoidWithPosition`

Currently, these interfaces do not extend `Requirable`, which is part of the command framework and which is Strongback's much simpler version of a WPILib "subsystem". Basically, when a command is submitted, the command scheduler looks through all existing-and-still-running commands to see if they require the same `Requirable` object as the new command; any existing commands that do are cancelled and removed. IOW, if a new command requires object A, then any existing-but-still-running commands that also require object A are cancelled, ensuring that two concurrently running commands don't compete with each other.

But imagine a simple robot with a single motor and encoder that drives an crane-like arm:

```
Motor armMotor = ...
AngleSensor armAngle = ...
```

Currently, if one wants to write multiple commands that use the motor, they have to define a separate `Requirable` object that these multiple commands can both require. Sure, it's easy to do:

```
static final Requirable ARM_LOCK = new Requirable(){};
```

Then we can define our commands
```
public class RaiseArm implements Command {
    private final Motor armMotor;
    public RaiseArm( Motor armMotor ) {
        super(ARM_LOCK);
        this.armMotor = armMotor;
    }
    ...
}

public class StowArm implements Command {
    private final Motor armMotor;
    public StowArm( Motor armMotor ) {
        super(ARM_LOCK);
        this.armMotor = armMotor;
    }
    ...
}
```

But we can make it slightly easier by making `Motor` and `Solenoid` (and by extension their sub-interfaces) extend `Requirable` so that the motor can be used as the requirable. This means we don't need that `ARM_LOCK` object at all, and our commands are slightly simpler:

```
public class RaiseArm implements Command {
    private final Motor armMotor;
    public RaiseArm( Motor armMotor ) {
        super(armMotor);
        this.armMotor = armMotor;
    }
    ...
}

public class StowArm implements Command {
    private final Motor armMotor;
    public StowArm( Motor armMotor ) {
        super(armMotor);
        this.armMotor = armMotor;
    }
    ...
}
```

Thoughts?

that the scheduler it is not possible to write multiple commands to

WawerOS, 2015-09-13:
Makes sense to me, in most situations you will be declaring a requirable.

rhauch, 2015-09-14:
Completed with PR #12.

closed 2015-09-14 by rhauch