strongback/strongback-java #25: label:Enhacement WPILib features missing in strongback.

granjef3, 2015-10-23:
So far, strongback seems to be capable of almost everything that WPILib only robot projects can do. However, there are a few cases where one would need to reach directly into WPILib with the current implementation to achieve, mainly control of control system components through CAN

1. Easy to use PID Loops. 
* PID on the Talon SRX
* CAN control of Jaguar (Not that important since Jaguars are now deprecated)
* Control system component status
* Vision/Camera related code
* Missing HAL things like 
 * Serial Port
 * Analog Input/Outputs
 * Digital Input/Outputs
 * Hardware Counter
 * Hardware Interrupts
* NetworkTables/General networking features

1 would probably be best solved by a PIDCommand subclass similar to WPILib. This one is somewhat challenging since 

2,3, and 4 are just extending their respective classes to add the relevant methods. 4 specifically is off to a good start with the PDP, but other things like the PCM and the roboRio voltage sensor

There is also no way at all to communicate with the PCM through strongback. (the PCM is named as wpilib.Compressor for some odd reason in WPILib, but strongback should probably just refer to it as the PCM to avoid confusion.

5. I have never used WPILib's camera/vision facilities, but looking through the code, they do make use of threads to receive images, which would conflict with the Executor idea of only two threads.

6. None of these items use threading, so converting their interfaces to use strongback paradigms would be relatively easy.

I'd be willing to help out on 1, 2, 4, and 6, but I don't have hardware to test 3 and 5.

7. Might be out of the scope of strongback, but Open-RIO/Toast provides Open-RIO/NetworkDelegate for splitting packets between Toast modules that use networks. The toast module system is interesting, but I don't know how it fits together with strongback

*I had to s/stronghold/strongback/ way too many times while writing this issue :joy:*

rhauch, 2015-10-23:
@granjef3, welcome, and these all sound great and along similar lines to what we were planning. Feel free to propose some improvements in these areas. We would prefer keeping PRs small, tho.

BTW, we've been using the 'Strongback' name since summer, and recently I've been having trouble typing 'Stronghold'. Too funny.

granjef3, 2015-10-23:
I don't get as much time to play with our 2015 robot anymore now that we are done with offseason competitions, but I'll try to dust it off sometime next week and port the code over to strongback. In the process I'll have to solve items 1,2, and part of 6 (PCM module)

As far as more support for different sensors (we used the ADXRS453 gyro, and the NavX is another popular choice) would you want things like this to be supported directly by strongback or through a module system similar to Toast. (Toast is a great concept, but the design philosophy of strongback appeals to me much more]

Also, In the future, do you prefer things like this feature request to be separate issues vs. one group of related requests?

rhauch, 2015-10-23:
@granjef3, I would prefer to have separate relatively focused issues, so that each issue is easily tackled, reviewed, and merged independently of the other ongoing work. Smaller issues generally get fixed faster, too, and they often make it easier for contributors to distribute the effort while working independently. For example, each of your major items could easily be separate issues. Some of the larger ones might even benefit from breaking into smaller issues, too. So if you create new smaller issues, please edit the above text and comments to reference the new issue. We can then leave this issue open for a while as a longer-lived discussion.

BTW, regarding NetworkTables: we know there are lots of issues with them, but despite that they are still used. So we will want a story here, and my gut feeling is that 1) if we can provide some benefit on top of NetworkTables to make it easier to use, then we should do that; and 2) if we can do that and *also* use the same API for other protocols then that's even better. However, I'd rather not create yet-another-network-tables-replacement (YANTR), so reusing others is fine with me.

There is a fair amount of hardware that we don't yet support, but which should be relatively easy to do -- especially by teams that have *used* that hardware. I'm not convinced we need pluggable modules, though, because they add a ton of complexity for contributors and users. Teams can *always* create custom classes that implement existing Strongback hardware interfaces; they just wouldn't use the built-in Strongback hardware factory methods. So extensibility is quite natural, though we'd love for these kinds of extensions to be added to Strongback via pull requests!

rhauch, 2015-10-23:
@granjef3, I've added you as a collaborator on the repository, so feel free to assign any issues to yourself. That's the best way for everyone to know that someone is working on the issue.

Zabot, 2015-10-24:
I started on some code for a software `RIO` that handles channel allocation the other day. I'll make an issue for it tomorrow, but that can probably be expanded to include the `PCM` and any integrated sensors.

rhauch, 2015-10-27:
@granjef3, I just added issue #29 for adding PID control classes (including a command that uses the PID control), plus pull-request #30 that implements the functionality. I updated the description above to reference the issue in the relevant places. Please take a look.

rhauch, 2015-10-27:
> 4. Control system component status
@granjef3, can you explain a bit more about what this means? Is some of that covered in #30?

granjef3, 2015-10-27:
@rhauch I was referring to the faults/sticky faults of the various control system components that communicate over CAN. Currently, `PneumaticsModule` supports sticky faults through the `compressorFaults()` and `compressorStickyFaults()` functions, but a similar feature does not exist for `PowerPanel`. When faults are added to `PowerPanel`, should we create `powerPanelFaults()` and `powerPanelStickyFaults()`, or create a `Fault` interface to be used between all control system components that have sticky faults. (PCM, PDB, Talon SRX, Jaguar) This `Fault` interface could possibly be useful for tracking custom robot components as well. `Fault` could extend the `DataRecordable` interface from #30, allowing one to use `DataRecorder.register()` to record all faults during robot operation.

rhauch, 2015-10-27:
@granjef3, the `PneumaticsModule.Faults` interface right now if very specific to the `PneumaticsModule`. Yes, we could add an analogous interface to `PowerPanel`, but it's almost certainly going to be specific to that component; if we do this, then it should be a separate issue ... can you log that, and do you want to work on it?

As for faults being recorded, I think faults would better be tracked under the `EventRecorder` because they are likely to be very infrequent. That means that whenever a fault occurs, we'd want to record the fault as a new event via `Strongback.eventRecorder().record(...)`. So no registration is required. But speaking of knowing when a fault occurs, does WPILib have a *generic* way to be called when those or any other faults occur? If not, we probably have to resort to polling via `SwitchListener`, and it sure would be better to do that with one low-level WPILib fault mechanism than multiple higher-level ones.

granjef3, 2015-10-27:
@rhauch see #31. WPILib does not have a generic fault tracking system, or any at all. You must poll the control system components yourself to keep track of their faults.

rhauch, 2016-10-28:
@granjef3 We're getting ready to make changes for the 2017 Beta test program, and any changes will go into the 2.x line of releases. We'd welcome pull requests for any of these improvements!