strongback/strongback-java #31: Control System Fault tracking

granjef3, 2015-10-27:
Control system components that communicate over CAN such as the PDB, PCM, TalonSRX/Jaguar all have faults and sticky faults. There is no central polling mechanism implemented by WPILib, so we should use `EventRecorder` in each component to track faults.

rhauch, 2015-10-27:
We probably want to make recording these faults optional, since the extra logic to poll all of these faults will end up being not trivial, and people may want to turn them off if they can be accessed elsewhere.

rhauch, 2015-10-27:
The `PneumaticsModule` interface already defines a pattern for named faults, how they are sticky, and how the sticky faults are cleared. Other components will have different types of faults, but hopefully we can reuse the same pattern.

granjef3, 2015-10-27:
The idea of an actual `hasFaults` interface seems useful to me... you don't
have to remember that `PneumaticModule` uses `compressorFaults()`, you have
a common interface and you just can call `faults()` and `stickyFaults()` on
any class that implements `hasFaults`

granjef3, 2015-10-27:
@rhauch Can you assign this issue to me? I don't have privileges to do so.

rhauch, 2015-10-27:
@granjef3, it is now assigned to you. Thanks!

granjef3, 2015-10-28:
@rhauch Just curious, how do you setup your workspace in eclipse for strongback development? I just set my workspace to the `strongback-java` directory and imported all the child folders as projects, but I'm not sure if this is the correct way.

rhauch, 2015-10-28:
@granjef3, yes, that's what we do. Also, the `ide-config` folder contains some Java templates, formatter, and cleanup preferences that you can import to help standardize the auto-formatting in Eclipse.

rhauch, 2015-10-28:
@granjef3, well, I just create a new workspace in the `strongback-java` directory, and then import the existing projects. Also, the `ide-config` folder contains some Java templates, formatter, and cleanup preferences that you can import to help standardize the auto-formatting in Eclipse.

rhauch, 2015-11-15:
@granjef3, any progress on this issue? If you have code that is not done, you can still create a pull-request and just denote in the PR description that it's incomplete.

granjef3, 2015-11-15:
I've been busy with my birthday celebrations. I will try to work on the code this week, sorry about the delay.

rhauch, 2015-11-15:
@granjef3, well Happy Birthday! We'd like to release 1.1.0 later this week, after #40 (expose PID control for Talon SRX) is done.