strongback/strongback-java #36: Ant test randomly fails/passes on Mac OSX

granjef3, 2015-11-02:
Ant has been very buggy for me, it usually takes ~7 executions of `ant test` for it to actually succeed.
I'm on Mac OSX 10.10.3, Java version 1.8.0_66, ant version 1.9.5. Build will either fail at `compile-tests` or `compile`. Ant was installed with homebrew.

No idea if this is a configuration problem, or an problem with strongback. I dealt with a similar issue on java 1.8.0_60 during our offseason where code deployment would only succeed 1/5th of the time

Error output below:

```
compile-tests:
     [echo] Compiling against WPILib 0.1.0.201502241928 installed at /Users/matthew/Code/Projects/strongback-java/libs/wpilib
    [mkdir] Created dir: /Users/matthew/Code/Projects/strongback-java/strongback-tools/build/test-classes
    [javac] Compiling 2 source files to /Users/matthew/Code/Projects/strongback-java/strongback-tools/build/test-classes
    [javac] An exception has occurred in the compiler (1.8.0_66). Please file a bug at the Java Bug Database (http://bugreport.java.com/bugreport/) after checking the database for duplicates. Include your program and the following diagnostic in your report.  Thank you.
    [javac] java.lang.NullPointerException
    [javac] 	at com.sun.tools.javac.file.ZipFileIndexArchive$ZipFileIndexFileObject.<init>(ZipFileIndexArchive.java:121)
    [javac] 	at com.sun.tools.javac.file.ZipFileIndexArchive.getFileObject(ZipFileIndexArchive.java:76)
    [javac] 	at com.sun.tools.javac.file.JavacFileManager.listArchive(JavacFileManager.java:308)
    [javac] 	at com.sun.tools.javac.file.JavacFileManager.listContainer(JavacFileManager.java:355)
    [javac] 	at com.sun.tools.javac.file.JavacFileManager.list(JavacFileManager.java:624)
    [javac] 	at com.sun.tools.javac.jvm.ClassReader.fillIn(ClassReader.java:2750)
    [javac] 	at com.sun.tools.javac.jvm.ClassReader.complete(ClassReader.java:2446)
    [javac] 	at com.sun.tools.javac.jvm.ClassReader.access$000(ClassReader.java:76)
    [javac] 	at com.sun.tools.javac.jvm.ClassReader$1.complete(ClassReader.java:240)
    [javac] 	at com.sun.tools.javac.code.Symbol.complete(Symbol.java:574)
    [javac] 	at com.sun.tools.javac.code.Symbol$PackageSymbol.members(Symbol.java:858)
    [javac] 	at com.sun.tools.javac.comp.MemberEnter.importAll(MemberEnter.java:156)
    [javac] 	at com.sun.tools.javac.comp.MemberEnter.visitTopLevel(MemberEnter.java:525)
    [javac] 	at com.sun.tools.javac.tree.JCTree$JCCompilationUnit.accept(JCTree.java:518)
    [javac] 	at com.sun.tools.javac.comp.MemberEnter.memberEnter(MemberEnter.java:437)
    [javac] 	at com.sun.tools.javac.comp.MemberEnter.complete(MemberEnter.java:1038)
    [javac] 	at com.sun.tools.javac.code.Symbol.complete(Symbol.java:574)
    [javac] 	at com.sun.tools.javac.code.Symbol$ClassSymbol.complete(Symbol.java:1037)
    [javac] 	at com.sun.tools.javac.comp.Enter.complete(Enter.java:493)
    [javac] 	at com.sun.tools.javac.comp.Enter.main(Enter.java:471)
    [javac] 	at com.sun.tools.javac.main.JavaCompiler.enterTrees(JavaCompiler.java:982)
    [javac] 	at com.sun.tools.javac.main.JavaCompiler.compile(JavaCompiler.java:857)
    [javac] 	at com.sun.tools.javac.main.Main.compile(Main.java:523)
    [javac] 	at com.sun.tools.javac.main.Main.compile(Main.java:381)
    [javac] 	at com.sun.tools.javac.main.Main.compile(Main.java:370)
    [javac] 	at com.sun.tools.javac.main.Main.compile(Main.java:361)
    [javac] 	at com.sun.tools.javac.Main.compile(Main.java:56)
    [javac] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [javac] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    [javac] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [javac] 	at java.lang.reflect.Method.invoke(Method.java:497)
    [javac] 	at org.apache.tools.ant.taskdefs.compilers.Javac13.execute(Javac13.java:57)
    [javac] 	at org.apache.tools.ant.taskdefs.Javac.compile(Javac.java:1160)
    [javac] 	at org.apache.tools.ant.taskdefs.Javac.execute(Javac.java:936)
    [javac] 	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:293)
    [javac] 	at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
    [javac] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [javac] 	at java.lang.reflect.Method.invoke(Method.java:497)
    [javac] 	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
    [javac] 	at org.apache.tools.ant.Task.perform(Task.java:348)
    [javac] 	at org.apache.tools.ant.Target.execute(Target.java:435)
    [javac] 	at org.apache.tools.ant.Target.performTasks(Target.java:456)
    [javac] 	at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1405)
    [javac] 	at org.apache.tools.ant.helper.SingleCheckExecutor.executeTargets(SingleCheckExecutor.java:38)
    [javac] 	at org.apache.tools.ant.Project.executeTargets(Project.java:1260)
    [javac] 	at org.apache.tools.ant.taskdefs.Ant.execute(Ant.java:441)
    [javac] 	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:293)
    [javac] 	at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
    [javac] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [javac] 	at java.lang.reflect.Method.invoke(Method.java:497)
    [javac] 	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
    [javac] 	at org.apache.tools.ant.Task.perform(Task.java:348)
    [javac] 	at org.apache.tools.ant.Target.execute(Target.java:435)
    [javac] 	at org.apache.tools.ant.Target.performTasks(Target.java:456)
    [javac] 	at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1405)
    [javac] 	at org.apache.tools.ant.helper.SingleCheckExecutor.executeTargets(SingleCheckExecutor.java:38)
[... 93 lines cut ...]
    [javac] 	at org.apache.tools.ant.taskdefs.Ant.execute(Ant.java:441)
    [javac] 	at org.apache.tools.ant.taskdefs.CallTarget.execute(CallTarget.java:105)
    [javac] 	at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:293)
    [javac] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
    [javac] 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
    [javac] 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
    [javac] 	at java.lang.reflect.Method.invoke(Method.java:497)
    [javac] 	at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:106)
    [javac] 	at org.apache.tools.ant.Task.perform(Task.java:348)
    [javac] 	at org.apache.tools.ant.Target.execute(Target.java:435)
    [javac] 	at org.apache.tools.ant.Target.performTasks(Target.java:456)
    [javac] 	at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1405)
    [javac] 	at org.apache.tools.ant.Project.executeTarget(Project.java:1376)
    [javac] 	at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
    [javac] 	at org.apache.tools.ant.Project.executeTargets(Project.java:1260)
    [javac] 	at org.apache.tools.ant.Main.runBuild(Main.java:853)
    [javac] 	at org.apache.tools.ant.Main.startAnt(Main.java:235)
    [javac] 	at org.apache.tools.ant.launch.Launcher.run(Launcher.java:285)
    [javac] 	at org.apache.tools.ant.launch.Launcher.main(Launcher.java:112)

BUILD FAILED
/Users/matthew/Code/Projects/strongback-java/build.xml:57: The following error occurred while executing this line:
/Users/matthew/Code/Projects/strongback-java/dependencies.xml:17: The following error occurred while executing this line:
/Users/matthew/Code/Projects/strongback-java/build-common.xml:69: Compile failed; see the compiler error output for details.
```

rhauch, 2015-11-02:
I was on OS X 10.10.4 and as of late last week am now on 10.11.1. I'm currently using Ant 1.9.2 installed via MacPorts (installed at least a year ago) and Oracle JDK Java 1.8.0_45 with the following specifics:

    java version "1.8.0_45"
    Java(TM) SE Runtime Environment (build 1.8.0_45-b14)
    Java HotSpot(TM) 64-Bit Server VM (build 25.45-b02, mixed mode)

I've never seen the error you've come across. A few sites suggest that maybe it's related to constrained memory. Do you have `JAVA_OPTS` or `ANT_OPTS` set to use a smaller memory limit? These are not currently set on my machine. Perhaps try `ant -v compile` to see if that includes anything useful.

granjef3, 2015-11-03:
I don't have `ANT_OPTS` or `JAVA_OPTS` set on my machine.

Here is my full output using `ant -v` for both `test` and `compile`. Procedure was doing a fresh `git clone

rhauch, 2015-11-03:
@granjef3, please try removing [line 38](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L38) from `build-common.xml`. It is incorrect, and perhaps that empty variable is resulting in the NPE in the ZIP file code.

rhauch, 2015-11-03:
@granjef3, I've installed JDK 1.8.0_66 (same version you're using), and I'm not seeing the error. I wonder if this is a problem with your version of Ant (1.9.5). I downloaded Ant 1.9.5 from the [Apache archive](http://archive.apache.org/dist/ant/binaries/), installed it in a temp directory, and added it to my path:

    $ export PATH=~/Desktop/apache-ant-1.9.5/bin:$PATH
    $ ant clean
    ...
    BUILD SUCCESSFUL
    Total time: 0 seconds
    $ ant compile
    ...
    BUILD SUCCESSFUL
    Total time: 2 seconds

I then tried Ant 1.9.6 and got the same result. BTW, I did not remove line 38 as mentioned above, and my code is exactly what's currently in `master`.

I'm not sure what to suggest. The fact that you had similar problems last year before you tried using Strongback suggests something is wrong with your local system? Perhaps check or re-install the latest WPILib version from 2015? Or perhaps consider downloading and installing Ant 1.9.6 (at least in a temporary area)?

rhauch, 2015-11-03:
@granjef3, I found a couple of mistakes in the classpath definition (see #37), although I was not testing with those. You may want to get the latest and try again.

granjef3, 2015-11-04:
Yeah, the build fixes don't change the error... Hmm. I'll try messing with my local configuration to see what that does.

rhauch, 2015-11-09:
@granjef3, any luck so far?

granjef3, 2015-11-09:
Sorry for the lack of updates, been very busy with school. I'll try installing macports version of ant and see what that does, hopefully it fixes it. If I can't get that working, I'll try running ant in vagrant to get around my local environment being so borked.

granjef3, 2015-11-09:
Nope, ant from macports doesn't change anything :/ will try reinstalling jdk next.

rhauch, 2016-10-28:
@granjef3 Any progress on figuring this out? I still haven't ever seen this on my MBP.