strongback/strongback-java #47: ant test fails for Ubuntu 14.04

jaimeyu, 2015-11-28:
Hey, pretty stock ubuntu 14.04 VM with FRC toolchain and I'm trying to build Strongback from scratch. I get test failures (looks like junit is not in the classpath).

Raw logs at https://gist.github.com/jaimeyu/010dc9328c25e1ecb1b3
If you want the VM, let me know, I will privately share it. 

(I also get the same issue when I run 'ant test' for newly made strongback.sh projects).

rhauch, 2015-12-10:
@Zabot, can you try to replicate this on Ubuntu?

rhauch, 2016-10-28:
This looks similar to the errors reported in #49; I think Travis uses Ubuntu in their images. Funny thing is that our Travis builds are working. Would appreciate anyone on Ubuntu giving this a whirl to see if they can identify the problem or a fix.

AGausmann, 2017-01-17:
This is definitely related to #49 and I am able to reproduce a similar problem. I get a `NoClassDefFoundError` when it tries to access classes from hamcrest. There must be some inconsistency in how/when dependencies are managed between systems. Here is my output when running `ant test` for reference: [test.txt](https://github.com/strongback/strongback-java/files/710062/test.txt)

WawerOS, 2017-01-17:
I found that adding the jars in libs/test to CLASSPATH fixed the issue. I don't know how to implement it in ant though.

AGausmann, 2017-01-18:
That's what I'm assuming. We can probably figure out how to do that in Ant, or have Ant download them to a designated place instead of hosting them in the repository. Alternatively, if we migrate to Maven builds ( #92 ), that will likely also fix our issue.

rhauch, 2017-01-18:
> I found that adding the jars in libs/test to CLASSPATH fixed the issue. I don't know how to implement it in ant though.

The Ant build file is already doing this, or at least it is supposed to be doing this. This currently works on OS X on Ubuntu 12.04 (as evidenced by Travis-CI and, for example, [this recent pull request](https://s3.amazonaws.com/archive.travis-ci.org/jobs/192569552/log.txt)). Note that I and Travis-CI are using Ant 1.10.0 (mostly because of recent fixes to Ant's HTTP GET utility, which we need to download 3rd party libraries).

It works this way: the Ant build file (actually `build-common.xml`) is defining a classpath named `test.class.path` at [line 46](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L46) comprised of the normal `compile.class.path` (defined [on line 37](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L37)) plus the `*.jar` files (except `*-sources.jar`) files in the `${strongback.dir}/libs/test` directory or subdirectories, then using this classpath when [compiling the tests](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L93) and [running the tests](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L114). 

Bottom line is that this definitely varies on different systems. @jaimeyu is running Ubuntu 14.04, but what OS are you running, @AGausmann and @WawerOS? And what version of Ant are you all running? Do you each have all of the following files are in the `lib/test` directory?
* fest-assert-1.4-sources.jar
* fest-util-1.1.6-sources.jar
* hamcrest-core-1.3-sources.jar
* junit-4.11-sources.jar
* metrics-core-3.1.0-sources.jar
* fest-assert-1.4.jar
* fest-util-1.1.6.jar
* hamcrest-core-1.3.jar
* junit-4.11.jar
* metrics-core-3.1.0.jar

One thing you might try if you haven't already is to run Ant with verbose mode via `ant -v clean test`.

rhauch, 2017-01-18:
BTW, #49 had this in the output:

> Trying to override old definition of task classloader

@AGausmann, you didn't include enough output to see whether you're getting the same thing. And @jaimeyu's file didn't have it either.

AGausmann, 2017-01-18:
I am not getting that message in Strongback itself, but I have seen it in WPILib robot projects that are using Strongback.

Here is `ant -v clean test` output (sorry, apparently the previous output got screwed up when I piped it into a file.)

rhauch, 2017-01-18:
Ok, I may have found the problem but I'd like someone to try a potential fix first. Here's [ine 93-96 of build-common.xml](https://github.com/strongback/strongback-java/blob/master/build-common.xml#L93):

            <classpath refid="test.class.path" />
            <classpath> 
                <pathelement location="${project.test.output.dir}"/>
            </classpath>

Note how there are two `<classpath>` elements? I think these lines should be changed to:

            <classpath> 
                <path refid="test.class.path" />
                <pathelement location="${project.test.output.dir}"/>
            </classpath>

@AGausmann, @WawerOS, or @jaimeyu, would one of you be willing to make this change locally and try it? If that works, can you submit a PR?

AGausmann, 2017-01-18:
Just tried, and it didn't work. BTW, there was no difference in the output/result when I substituted `test.class.path` for `test.run.class.path` in the `<classpath>` tag.

WawerOS, 2017-01-19:
I am running Xubuntu version 16.10 and ant version 1.10, here is the [log](https://gist.github.com/WawerOS/62b5f87f3f609738139fbc4ba98c48ef)from ant -v test. I tried both variations of the buidld file and they didn't work.