strongback/strongback-java #48: jar packaging fails

forbesk, 2015-11-30:
Hi! I just started messing with Strongback (great so far). I noticed an issue today when we first tried getting some basic code up and running. The `FRCUserProgram.jar` created in `ant/strongback/build.xml` was missing the WPIlib, Strongback, and NetworkTables jars. The build completed fine and attempted to run on the roboRIO, but, as expected, it could not find the `RobotBase` class.

I got it running by changing the following 
```java
...
<target name="jar" depends="compile">
	<echo>Making jar ${dist.jar}.</echo>
	<mkdir dir="${dist.dir}" />
	<mkdir dir="${build.jars}" />

	<jar destfile="${dist.jar}" update="false">
		<manifest>
			<attribute name="Main-Class" value="edu.wpi.first.wpilibj.RobotBase"/>
			<attribute name="Robot-Class" value="${robot.class}"/>
			<attribute name="Class-Path" value="."/>
		</manifest>

		<fileset dir="${build.dir}" includes="**/*.class"/>

		<zipgroupfileset dir="${build.jars}">
			<include name="**/*.jar" />
		</zipgroupfileset>
	</jar>

	<echo>Copying jars from ${strongback.classpath} to ${build.jars}.</echo>
	<copy todir="${build.jars}" flatten="true">
		<file file="${dist.jar}"/>
		<path>
			<pathelement path="${strongback.classpath}"/>
		</path>
	</copy>
</target>
...
```

to this:
```java
...
<target name="jar" depends="compile">
	<echo>Making jar ${dist.jar}.</echo>
	<mkdir dir="${dist.dir}" />
	<mkdir dir="${build.jars}" />

	<echo>Copying jars from ${strongback.classpath} to ${build.jars}.</echo>
	<copy todir="${build.jars}" flatten="true">
		<path>
			<pathelement path="${strongback.classpath}"/>
		</path>
	</copy>

	<jar destfile="${dist.jar}" update="false">
		<manifest>
			<attribute name="Main-Class" value="edu.wpi.first.wpilibj.RobotBase"/>
			<attribute name="Robot-Class" value="${robot.class}"/>
			<attribute name="Class-Path" value="."/>
		</manifest>

		<fileset dir="${build.dir}" includes="**/*.class"/>

		<zipgroupfileset dir="${build.jars}">
			<include name="**/*.jar" />
		</zipgroupfileset>
	</jar>
</target>
...
```

I wouldn't even call myself an acquaintance to ant, but moving the `<copy ...>...</copy>` above the `<jar ...>...</jar>` and dropping `<file file="${dist.jar}"/>` created a correctly packaged `FRCUserProgram.jar` that ran on hardware.

rhauch, 2015-12-02:
Any chance you can submit a pull request with the change? If not, I'll try to find time to fix it in the next few days.

rhauch, 2015-12-10:
@forbesk, thanks for the bug report and suggested fix!

PR #50 (2015-12-10): Corrected the jar task in the build.xml that Strongback uses to build projects
Per #48, the "jar" task in the `build.xml` deployed and used to build projects was incorrect, and this PR fixes it per the suggestion in #48.

Fixes #48
closed 2015-12-10 by rhauch