strongback/strongback-java #49: Travis.yml incorrectly declares build&test as successful

jaimeyu, 2015-12-02:
Visited https://travis-ci.org/strongback/strongback-java-examples and noticed that it doesn't look like the travis.yml is correct. It doesn't seem to build the project nor test it. Can you upload a working example?

I ran this on my local VM and also experience the same issues with openjdk8 & ant (Eclipse works fine).

Snippet from the Travis logs. 
compile:
[mkdir] Created dir: /home/travis/build/strongback/strongback-java-examples/minimal-arcade-drive/build
[javac] Compiling 1 source file to /home/travis/build/strongback/strongback-java-examples/minimal-arcade-drive/build
[subant] Failure for target 'null' of: /home/travis/build/strongback/strongback-java-examples/minimal-arcade-drive/build.xml
[subant] The following error occurred while executing this line:
[subant] /home/travis/strongback/java/ant/build.xml:48: copy doesn't support the "quiet" attribute
Trying to override old definition of task classloader
compile:
[mkdir] Created dir: /home/travis/build/strongback/strongback-java-examples/simple-arcade-drive/build
[javac] Compiling 2 source files to /home/travis/build/strongback/strongback-java-examples/simple-arcade-drive/build
[subant] Failure for target 'null' of: /home/travis/build/strongback/strongback-java-examples/simple-arcade-drive/build.xml
[subant] The following error occurred while executing this line:
[subant] /home/travis/strongback/java/ant/build.xml:48: copy doesn't support the "quiet" attribute

rhauch, 2015-12-02:
What version of Ant do you have locally?

jaimeyu, 2015-12-02:
Apache Ant(TM) version 1.9.3 compiled on April 8 2014
Linux DeathStar 3.19.0-33-generic #38~14.04.1-Ubuntu SMP Fri Nov 6 18:17:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
$ java -version
openjdk version "1.8.0_45-internal"
OpenJDK Runtime Environment (build 1.8.0_45-internal-b14)
OpenJDK 64-Bit Server VM (build 25.45-b02, mixed mode)

jaimeyu, 2015-12-02:
Here is the output from my workstation:

# jaimeyu at DeathStar in ~/Github/strongback-java-examples on git:master o [23:46:51]
$ ant
Buildfile: /home/jaimeyu/Github/strongback-java-examples/build.xml

compile:
Trying to override old definition of task classloader

compile:
    [mkdir] Created dir: /home/jaimeyu/Github/strongback-java-examples/minimal-arcade-drive/build
    [javac] Compiling 1 source file to /home/jaimeyu/Github/strongback-java-examples/minimal-arcade-drive/build
Trying to override old definition of task classloader

compile:
    [mkdir] Created dir: /home/jaimeyu/Github/strongback-java-examples/simple-arcade-drive/build
    [javac] Compiling 2 source files to /home/jaimeyu/Github/strongback-java-examples/simple-arcade-drive/build

BUILD SUCCESSFUL
Total time: 1 second

# jaimeyu at DeathStar in ~/Github/strongback-java-examples on git:master o [23:46:53]
$ ant test
Buildfile: /home/jaimeyu/Github/strongback-java-examples/build.xml

test:
Trying to override old definition of task classloader

compile:

if-tests:

compile-tests:

test:
Trying to override old definition of task classloader

compile:

if-tests:

compile-tests:
    [mkdir] Created dir: /home/jaimeyu/Github/strongback-java-examples/simple-arcade-drive/tests/classes
    [javac] Compiling 1 source file to /home/jaimeyu/Github/strongback-java-examples/simple-arcade-drive/tests/classes

test:
     [echo] [run-tests] Running all unit tests...
    [junit] Testsuite: org.strongback.example.robot.TimedDriveCommandTest
    [junit] Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 0 sec
    [junit] 
    [junit] 	Caused an ERROR
    [junit] junit/framework/JUnit4TestAdapterCache
    [junit] java.lang.NoClassDefFoundError: junit/framework/JUnit4TestAdapterCache
    [junit] 	at java.lang.ClassLoader.defineClass1(Native Method)
    [junit] 	at java.lang.ClassLoader.defineClass(ClassLoader.java:760)
    [junit] 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
    [junit] 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
    [junit] 	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
    [junit] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
    [junit] 	at java.security.AccessController.doPrivileged(Native Method)
    [junit] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
    [junit] 	at java.lang.Class.forName0(Native Method)
    [junit] 	at java.lang.Class.forName(Class.java:264)
    [junit] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
    [junit] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
    [junit] 
    [junit] Test org.strongback.example.robot.TimedDriveCommandTest FAILED
   [subant] Failure for target 'null' of: /home/jaimeyu/Github/strongback-java-examples/simple-arcade-drive/build.xml
   [subant] The following error occurred while executing this line:
   [subant] /home/jaimeyu/strongback/java/ant/build.xml:122: Test failure detected, check test results.

BUILD SUCCESSFUL
Total time: 1 second

rhauch, 2015-12-02:
What version of strongback do you have? Also, the output from your local build is not at all the same problem as shown on the output from the Travis build. I found some references on the internet to some problems with Ant 1.9.3 on some flavors of Linux that suggested Ant was not properly building the classpath, and that forking the JUnit task fixed the problem.

First, can you please verify that your username has read privilege for all the unit test libraries by running the following and pasting the response?

$  ll ~/strongback/java/lib-tests/

If the libraries are there and they are globally readable, then please try again with `ant -debug clean test`, and paste all (or better yet part) of the output. For example, the useful part is at the end of my build:

```
...
compile-tests:
    [mkdir] Created dir: /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes
    [javac] org/strongback/example/robot/TimedDriveCommandTest.java added as org/strongback/example/robot/TimedDriveCommandTest.class doesn't exist.
    [javac] Compiling 1 source file to /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes
    [javac] Using modern compiler
    [javac] Compilation arguments:
    [javac] '-d'
    [javac] '/Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes'
    [javac] '-classpath'
    [javac] '/Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes:/Users/rhauch/strongback/java/lib/strongback.jar:/Users/rhauch/strongback/java/lib-tests/fest-assert-1.4.jar:/Users/rhauch/strongback/java/lib-tests/fest-util-1.1.6.jar:/Users/rhauch/strongback/java/lib-tests/hamcrest-core-1.3.jar:/Users/rhauch/strongback/java/lib-tests/junit-4.11.jar:/Users/rhauch/strongback/java/lib-tests/metrics-core-3.1.0.jar:/Users/rhauch/strongback/java/lib-tests/strongback-testing.jar:/Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/build:/Users/rhauch/wpilib/java/current/lib/WPILib.jar:/Users/rhauch/wpilib/java/current/lib/NetworkTables.jar'
    [javac] '-sourcepath'
    [javac] '/Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/testsrc'
    [javac] '-g:none'
    [javac] 
    [javac] The ' characters around the executable and arguments are
    [javac] not part of the command.
    [javac] File to be compiled:
    [javac]     /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/testsrc/org/strongback/example/robot/TimedDriveCommandTest.java
     [copy] No sources found.
     [copy] No sources found.

test:
    [mkdir] Skipping /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes because it already exists.
    [mkdir] Skipping /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/tests/classes because it already exists.
     [echo] [run-tests] Running all unit tests...
    [junit] Using System properties {java.runtime.name=Java(TM) SE Runtime Environment, sun.boot.library.path=/Library/Java/JavaVirtualMachines/jdk1.8.0_66.jdk/Contents/Home/jre/lib, java.vm.version=25.66-b17, ant.library.dir=/opt/local/share/java/apache-ant/lib, gopherProxySet=false, java.vm.vendor=Oracle Corporation, java.vendor.url=http://java.oracle.com/, path.separator=:, java.vm.name=Java HotSpot(TM) 64-Bit Server VM, file.encoding.pkg=sun.io, user.country=US, sun.java.launcher=SUN_STANDARD, sun.os.patch.level=unknown, java.vm.specification.name=Java Virtual Machine Specification, user.dir=/Users/rhauch/dev/java/strongback/strongback-java-examples, java.runtime.version=1.8.0_66-b17, java.awt.graphicsenv=sun.awt.CGraphicsEnvironment, java.endorsed.dirs=/Library/Java/JavaVirtualMachines/jdk1.8.0_66.jdk/Contents/Home/jre/lib/endorsed, os.arch=x86_64, java.io.tmpdir=/var/folders/hd/z9xm6y312hs3h9h49np58h6m0000gn/T/, line.separator=
    [junit] , java.vm.specification.vendor=Oracle Corporation, os.name=Mac OS X, ant.home=/opt/local/share/java/apache-ant, sun.jnu.encoding=UTF-8, java.library.path=/Users/rhauch/Library/Java/Extensions:/Library/Java/Extensions:/Network/Library/Java/Extensions:/System/Library/Java/Extensions:/usr/lib/java:., java.specification.name=Java Platform API Specification, java.class.version=52.0, sun.management.compiler=HotSpot 64-Bit Tiered Compilers, os.version=10.11.1, http.nonProxyHosts=local|*.local|169.254/16|*.169.254/16, user.home=/Users/rhauch, user.timezone=America/Chicago, java.awt.printerjob=sun.lwawt.macosx.CPrinterJob, file.encoding=UTF-8, java.specification.version=1.8, java.class.path=/opt/local/share/java/apache-ant/lib/ant-launcher.jar:/opt/local/share/java/apache-ant/lib/ant-antlr.jar:/opt/local/share/java/apache-ant/lib/ant-apache-bcel.jar:/opt/local/share/java/apache-ant/lib/ant-apache-bsf.jar:/opt/local/share/java/apache-ant/lib/ant-apache-log4j.jar:/opt/local/share/java/apache-ant/lib/ant-apache-oro.jar:/opt/local/share/java/apache-ant/lib/ant-apache-regexp.jar:/opt/local/share/java/apache-ant/lib/ant-apache-resolver.jar:/opt/local/share/java/apache-ant/lib/ant-apache-xalan2.jar:/opt/local/share/java/apache-ant/lib/ant-commons-logging.jar:/opt/local/share/java/apache-ant/lib/ant-commons-net.jar:/opt/local/share/java/apache-ant/lib/ant-jai.jar:/opt/local/share/java/apache-ant/lib/ant-javamail.jar:/opt/local/share/java/apache-ant/lib/ant-jdepend.jar:/opt/local/share/java/apache-ant/lib/ant-jmf.jar:/opt/local/share/java/apache-ant/lib/ [... cut ...]
    [junit] Testcase: shouldStopWhenCancelled took 0 sec
   [subant] Exiting /Users/rhauch/dev/java/strongback/strongback-java-examples/simple-arcade-drive/build.xml.

BUILD SUCCESSFUL
Total time: 1 second
```

Notice how the `-classpath` line includes `/Users/rhauch/strongback/java/lib-tests/junit-4.11.jar`? That's what it should do, but if Ant is not working correctly on your machine, then per the aforementioned reference the classpath would not contain the `~/strongback/java/lib-tests/junit-4.11.jar` file.

rhauch, 2016-10-28:
@jaimeyu Any progress on figuring out the issue? Our Travis build is working (see [this example](https://travis-ci.org/strongback/strongback-java/builds/171267890)).