strongback/strongback-java #54: error importing libraries into eclipse

bussell, 2016-01-11:
We have tried to get Strongback working in eclipse on multiple Windows machines with the same failure. When we get to the point where we import Strongback.userlibraries there is a path error. It appears that there is a slash missing. I have not been able to identify where the path is set.

 Here is the relevant error:
 org.eclipse.jdt.internal.core.ClasspathEntry$Asser tionFailedException: Path for IClasspathEntry must be absolute: C:Usersjoe_bstrongback/java/lib/strongback.jar

 My username is joe_b on this machine, you will note the missing '/' before 'strongback'.

 We were able to get things working with ease on a Mac, but most of our team's machines are Windows.

I submit this issue on Chief Delphi in case others run into it as a reply to the introduction post at: http://www.chiefdelphi.com/forums/showpost.php?p=1521168&postcount=17

rhauch, 2016-01-11:
What are the contents of the `strongback.properties` file that is located in the Strongback installation directory?

rhauch, 2016-01-11:
@Zabot, can you check that [this line](https://github.com/strongback/strongback-java/blob/master/strongback-tools/src/org/strongback/tools/utils/FileUtils.java#L37) is properly handling values for `user.home` that do not end with a slash? That might be the case on Windows.

rhauch, 2016-01-11:
@bussell, you can try to correct the `/strongback/java/templates/Strongback.userlibraries.import.template` file to have the correct path, and then import again.

bussell, 2016-01-11:
Here is my strongback.properties contents:
#Thu, 07 Jan 2016 17:33:53 -0600
strongback.home=${user.home}/strongback
strongback.ant.dir=${strongback.home}/java/ant/
strongback.templates.dir=${strongback.home}/java/templates

wpilib.home=${user.home}/wpilib
wpilib.props=${wpilib.home}/wpilib.properties

strongback.version=1.1.0
build.date=2016-01-07

rhauch, 2016-01-11:
@bussell: Thanks for supplying your `strongback.properties` file; that helps narrow down the problem to likely be in the command line tool rather than the environment. 

I'm not convinced that workaround will solve all of the problems, but it should allow you to manually import the user library so that your projects can compile. If the problem is where I suspect, the `strongback` command line functionality will have problems on Windows, but you could always have your Mac users create the projects or add Strongback to them. Then when they share their code, you should get the correct project definitions, and each of your Eclipse workspaces should compile your robot project since each would have the correct user library.

We'll try to get this fixed in the next day or two, though.

bussell, 2016-01-11:
I tried to set my paths by hand and now the Import button in eclipse seems inert.  I reboot just in case (this is Windows after all), but to no avail.  I appreciate your time in looking into this issue.  I will review this again after you have a chance to check out the command line tool.

closed 2016-01-12 by rhauch

rhauch, 2016-01-12:
@bussell: I've found the problem (see #55 for the changes), and have released a new version of Strongback with the fix. I've updated the book, so please remove the existing `strongback` directory, then download and install the version 1.1.1. Everything else like your path should work as-is, and you can even verify the version with `strongback.bat version` (or `strongback.sh version` on OS X and Linux).

bussell, 2016-01-12:
I am attempting to test this, but the process of running the initial batch has left my Eclipse setup in disarray.  I removed the entire folder where Eclipse lived and reinstalled it, but I still get the following error:
org.eclipse.jdt.internal.core.ClasspathEntry$AssertionFailedException: Source attachment path 'C:Usersjoe_bstrongback/java/lib/strongback-sources.jar' for IClasspathEntry must be absolute
	at org.eclipse.jdt.core.JavaCore.newLibraryEntry(JavaCore.java:4721)
	at org.eclipse.jdt.internal.ui.wizards.buildpaths.CPListElement.newClasspathEntry(CPListElement.java:200)
	at org.eclipse.jdt.internal.ui.wizards.buildpaths.CPListElement.getClasspathEntry(CPListElement.java:167)
	at org.eclipse.jdt.internal.ui.wizards.buildpaths.CPListElement.equals(CPListElement.java:492)
	at java.util.ArrayList.indexOf(Unknown Source)

From you understanding of the original script could you help me find out where the bad path entry is coming from?

rhauch, 2016-01-12:
@bussell: are you sure you are using 1.1.1?

bussell, 2016-01-12:
This is an Eclipse fail that occurs prior to selecting the library to import.  Just hitting the Import button results in this error.

rhauch, 2016-01-12:
Please follow the instructions to create a temporary project.

bussell, 2016-01-12:
I verified that build.properties displays 1.1.1

bussell, 2016-01-12:
I do not understand your request to create a temp project.  I have run the strongback.bat file to create a new project and verified that the strongback references are not working in that project.  There is some artifact that is preventing the act of starting the import dialog.  This same error occurs on a students machine as well.  If I wanted to start fresh, what do I have to delete other than the Eclipse and Strongback folders?

bussell, 2016-01-12:
"strongback.bat version" yields a newline with no data.

rhauch, 2016-01-12:
Make sure that the `Strongback` user library doesn't exist in your workspace preferences, and if it does then remove it.

Then check the `C:\Users\joe_b\strongback\java\eclipse` directory: there should be a single file, and if that has corrupted paths then no matter what you do you will not be able to import it into Eclipse.

rhauch, 2016-01-12:
> I do not understand your request to create a temp project. 

See the "Adding Strongback to your project" section of the (Getting Started chapter)[https://strongback.gitbooks.io/using-strongback/content/getting_started.html].

> I have run the strongback.bat file to create a new project 

With the `-e` option? You can try adding a `-D` option, which will output some debug messages. I wouldn't mind seeing that output.

> There is some artifact that is preventing the act of starting the import dialog. This same error occurs on a students machine as well. 

Again, check the contents of the `C:\Users\joe_b\strongback\java\eclipse\Strongback.userlib` file.

> If I wanted to start fresh, what do I have to delete other than the Eclipse and Strongback folders?

Delete the `.metadata` folder inside your Eclipse workspace folder. You are sure you have the latest WPILib plugins, too? What version of Windows? I tested this on Windows 10, and I don't have access to anything older tonight.

rhauch, 2016-01-12:
@bussell wrote:

> I verified that build.properties displays 1.1.1

What `build.properties` file? There's only one in the Strongback installation directory, and it's inside an `ant` folder and won't have the Strongback version in it. There is a `strongback/strongback.properties` file that will have the version number in it.

bussell, 2016-01-12:
the userlibraries file was missing some slashes.  I removed the file and reran the batch.  The result was that the dialog appeared on my next attempt.  I will have to remember that to clean this up on the student machine I will have to remove strongback\java\eclipse\Strongback.userlibraries and rerun the project builder.

Thanks for your time and attention.  This trouble ticket is closed from my perspective.

rhauch, 2016-01-12:
@bussell:

The strange thing is that if you completely removed the `strongback` directory, then the first time you would have run `strongback.bat new-project` it would have created the correct `userlibraries` file. Subsequent runs shouldn't have changed it.

Regardless, I think most of that was leftover from your previous attempts. Glad you got it working and thanks for your patience -- it shouldn't have been that difficult.

bussell, 2016-01-12:
Clearly my initial cleanup was not as complete as I thought.  Looking back I think I removed eclipse twice, but ran the older version for the batch file.  In any case, the fix for string replacement got me to the next step.

rhauch, 2016-01-12:
Thanks again for logging this and working with me to verify your problem was indeed fixed. Best of luck to your team!