strongback/strongback-java #56: Tank Drive Cheesy Joysticks swapped?

niharmitra, 2016-01-16:
It appears according to the documentation, the first joystick is the throttle and the second is the wheel. But from my team's testing it was the other way around?

rhauch, 2016-01-18:
Is [this the method in TankDrive](https://github.com/strongback/strongback-java/blob/master/strongback/src/org/strongback/drive/TankDrive.java#L245) to which you're referring?

    /**
     * Provide "cheesy drive" steering using a steering wheel and throttle. This function lets you directly provide joystick
     * values from any source.
     *
     * @param throttle the value of the throttle; must be -1 to 1, inclusive
     * @param wheel the value of the steering wheel; must be -1 to 1, inclusive Negative values turn right; positive values turn
     *        left.
     * @param isQuickTurn true if the quick-turn button is pressed
     * @see <a href="https://github.com/Team254/FRC-2014/blob/master/src/com/team254/frc2014/CheesyDriveHelper.java">Team 254
     *      Cheesy Drive logic</a>
     */
    public void cheesy(double throttle, double wheel, boolean isQuickTurn) {
        ...
    }

Note that this was derived from [Team 254's CheesyDriveHelper](https://github.com/Team254/FRC-2014/blob/master/src/com/team254/frc2014/CheesyDriveHelper.java#L21), and our method's logic and even parameter order seems to follow that of Team 254's.

Can you double check that you've not made a mistake elsewhere in your code?

EricLiu2000, 2016-01-20:
Yes that is the method. I just double checked, and it does seem that passing it the wheel then throttle works. We can post our usage of the code if you would like.

rhauch, 2016-01-20:
@EricLiu2000: I've looked at the code again in more depth, and I'm surprised to hear that the inputs are indeed switched. The existing logic just won't work if the `throttle` and `wheel` variables are semantically switched. I've also verified that [Team 254's 2015 code](https://github.com/Team254/FRC-2015/blob/master/src/com/team254/frc2015/CheesyDriveHelper.java) is nearly identical to the 2014 code mentioned previously and very similar to our code. 

That suggests to me that there is still a problem elsewhere in the code (including Strongback) or in the mapping between the physical devices and the software components on your robot. If you post your code, we'll go through it to see if we can find anything and (if time allows) test it on one of our team's robots. However, I'd ask that you double-check the physical mapping as well. If you're using two separate devices, double check that the port numbers in the code match the physical devices. Or, if you're using a single device, make sure that you're using the correct joystick implementation for your hardware device -- different joysticks are notorious for having different mappings between the buttons/controls and axes.

niharmitra, 2016-01-21:
https://github.com/team8/jFreyja/tree/strongback is the code we were running. We double checked the physical mapping in the driver station software, our joysticks were mapped to the ports we expected. Note that our code currently compensates for the swap.

rhauch, 2017-01-15:
@niharmitra Okay, I've looked at the code again and I'm still not convinced the behavior is as you describe. If you look closely at the logic, you'll see that the `throttle` and `wheel` components are indeed used correctly. I've also compared the logic to that from Team 254's code from [2014](https://github.com/Team254/FRC-2014/blob/master/src/com/team254/frc2014/CheesyDriveHelper.java) and [2016](https://github.com/Team254/FRC-2016-Public/blob/master/src/com/team254/frc2016/CheesyDriveHelper.java), and our logic is identical to theirs.

Are you sure that you didn't accidentally reverse the inputs? That is by far the most likely cause.