strongback/strongback-java #58: Robot throws a FileNotFoundException when using Strongback

AGausmann, 2016-01-17:
```
Unhandled exception: java.lang.RuntimeException: java.io.FileNotFoundException: strongback-data-1.dat (Permission denied) at [org.strongback.FileDataWriter.openIfNeeded(FileDataWriter.java:58), org.strongback.FileDataWriter.<init>(FileDataWriter.java:50), org.strongback.Strongback$Configurator.createFileDataWriter(Strongback.java:133), org.strongback.Strongback$Configurator$$Lambda$13/14944660.apply(Unknown Source), org.strongback.DataRecorderDriver.lambda$start$41(DataRecorderDriver.java:47), org.strongback.DataRecorderDriver$$Lambda$20/25376898.apply(Unknown Source), java.util.concurrent.atomic.AtomicReference.getAndUpdate(AtomicReference.java:160), org.strongback.DataRecorderDriver.start(DataRecorderDriver.java:47), org.strongback.Strongback.doStart(Strongback.java:740), org.strongback.Strongback.start(Strongback.java:398), org.frc4931.robot.Robot.teleopInit(Unknown Source), edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:135), edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:241)]
```
This occurs when I call `Strongback.start()` in my robot's `teleopInit()` method. Removing the method call fixes the problem, but I assume Strongback is not fully initialized if I do so.

rhauch, 2016-01-20:
Try turning off data and event recording in Strongback's configuration. And, since the exception is due to permission problems, so it’d be awesome if you can try to debug the problem and see if the fix is a change to the directory structure or if a different path is needed this year.

AGausmann, 2016-01-21:
I found the problem. The working directory of our robot was the root... obviously an unprivileged user wouldn't have write access. I patched the problem by running this in `robotInit()`:
```
Strongback.configure()
                .recordDataToFile("/home/lvuser/")
                .recordEventsToFile("/home/lvuser/", 2097152); // 2 megabytes
```

rhauch, 2016-01-21:
I'm going to reopen this issue so that we change the *default* directories where these files are read. Currently, the *default* location is not writable on the robot, which is run from the root directory as user `lvuser`. However, the default really should be the home directory for this user, which we can find using system properties.

rhauch, 2016-12-19:
Closing after better documenting the `recordDataToFile(...)` and `recordEventsToFile(..)` methods in #85.

closed 2016-12-19 by rhauch