strongback/strongback-java #62: Unable to use SwitchReactor

bdavies01, 2016-02-09:
When we try to use SwitchReactor, it doesn't seem to call the command at all.

`protected SwitchReactor reactor = Strongback.switchReactor();`
We set up the SwitchReactor one time as: 
`reactor.onTriggered(input.getDriveStick().getTrigger(), () -> Strongback.submit(new ToggleGears(this)));`

The command itself has been verified to be working when being manually submitted to the Strongback queue, so the lambda function is not being called as far as we can tell. This did not work with a direct print to the console nor using a direct command with a lambda function. This failed both with MockFlightSticks that we wrote and forcefully triggered, and with real joysticks on the robot as well.

rhauch, 2016-02-09:
@bdavies01: is this the same as #60? If so, then this is a duplicate.

bdavies01, 2016-02-09:
Sorry, we didn't realize that `SwitchReactor` was of type `AsyncSwitchReactor` and assumed this was a different issue. Do you have an estimated fix-time or suggested work-around? Our current plan is to repeatedly check Joystick values during an update cycle.

granjef3, 2016-02-10:
My open pull request that references #60 fixes AsyncSwitchReactor

rhauch, 2016-02-10:
Closing as a duplicate of #60.

closed 2016-02-10 by rhauch