strongback/strongback-java #68: Expose CANTalon.setPosition(double) in TalonSRX

AGausmann, 2016-02-29:
It is currently impossible to truly zero the TalonSRX's encoder input if you are using the closed-loop control. Calling `talon.getEncoderInput().zero()

rhauch, 2016-02-29:
It's not possible because it is unnecessary.

When you work with a Talon SRX device in close-loop mode, you use a `TalonController` (which extends both `TalonSRX` and `PIDController`) and configure it such that the encoder device wired into the Talon SRX device is properly represented with a `Gyroscope`. Yes, the `Gyroscope` object can be zeroed, but when you call `zero()` it simply remembers the reading at that point and subtract it from all subsequent angles returned by the `Gyroscope`.

Then, to command the Talon SRX device to move the motor until the encoder reaches a designated angle using closed loop control, your code should call `TalonController.withTarget(value)` and supply the desired angle as would be read by the `Gyroscope` (which presumably you've previously zeroed). The implementation of this method is [here](), and it adds the zero point back in _then_ instructs the Talon SRX's close loop control.

The bottom line is that the _zero_ behavior is removed from the `Gyroscope` output _and_ added back in when setting the target for the close-loop control.

Does this make sense? Or did I misunderstand your statement?

AGausmann, 2016-02-29:
If I understand what you are saying, the `TalonController.withTarget(target)` uses the offset that the Talon's `Gyroscope` sets when `zero()` is called?

rhauch, 2016-02-29:
@AGausmann, yes. If you look at [the code](https://github.com/strongback/strongback-java/blob/master/strongback/src/org/strongback/hardware/HardwareTalonSRX.java#L96), you'll see that the zero value is added back in to the raw target value for the encoder before it is pushed down to the Talon SRX device.

AGausmann, 2016-02-29:
Okay, thanks.

closed 2016-02-29 by AGausmann