strongback/strongback-java #80: NavX support for Strongback

plato2000, 2016-08-03:
Our team uses the NavX MXP device. It consists of a suite of sensors including a 3 axis gyroscope, a 3 axis accelerometer, a magnetometer, and an altimeter. It has [its own library](https://github.com/kauailabs/navxmxp/tree/master/roborio/java) that makes getting this data from the sensor easy. I think the library also allows for some fancy calculations for distance traveled and some other stuff.

I would like to add support for this sensor to Strongback. However, I'm not sure of the best way to do this. I think the NavX is popular enough that it should be supported by Strongback (complete with mock sensor for unit tests), but not popular enough to force all Strongback users to download the library for it. Also, I don't know whether the best idea is to consider the NavX as separate sensors for the accelerometer and the gyroscope or to consider it as its own class with fields for the individual sensors. The link above has some examples of how the library is used if that helps in making the decision.

I would be glad to submit a pull request with support for the NavX if someone would tell me the best method of adding it to fit with Strongback's system.

rhauch, 2016-08-03:
> Our team uses the NavX MXP device. It consists of a suite of sensors including a 3 axis gyroscope, a 3 axis accelerometer, a magnetometer, and an altimeter.

I have no experience with this device, so the big question you have to answer is: what should the API look like for this device? Is it as simple as a composition of a 3-axis gyroscope, 3-axis accelerometer, compass, and distance sensor (altimeter)? If so, then we probably just want to create an interface that has getter methods for the other existing components. Not sure what that might be named, or where it should live, but I'm fine with proceeding (at least for now) with it in the `org.strongback.components` package.

Then, an implementation of this new interface needs to be added to the hardware package, and it could be named `HardwareNavX` and should implement the new interface. Then, a new factory method needs to be added to the `Hardware` class.

> It has its own library that makes getting this data from the sensor easy.

That's okay. The library appears to be [licensed under MIT open source license](https://github.com/kauailabs/navxmxp/blob/master/LICENSE.txt), so that means we can distribute it. It would be easier for Strongback to include the library so that it works if people want to use it, but if they don't they can always remove it (if they really care about the small space).

For now, the JARs and source JARs should be included in the `libs` directory of our codebase. We may want to eventually change the build to be able to download the libraries from the official location, but we can worry about that later.

> I think the library also allows for some fancy calculations for distance traveled and some other stuff.

Hopefully the interface for this device remains pretty simple, and any other functionality (like distance travelled) should be exposed through existing components (e.g., `DistanceSensor`).

This should be a pretty exciting addition, but it may take multiple iterations to settle on something that is generic and that reuses much of the existing API. Either way, how about trying to fit it in as I described above, submitting a pull request, and then we'll review it and see where it needs to go?

rhauch, 2016-10-26:
@plato2000 any progress on a contribution to add support for the NavX?

AGausmann, 2016-11-10:
I would be interested in implementing this in Strongback if no progress has been made. @plato2000 ?

plato2000, 2016-11-10:
@AGausmann Unfortunately, I have not had time with school and everything to make progress on this. I probably won't have time before the Christmas break just before the season. I have not made any progress on this yet. Feel free to implement it yourself if you can start by then.

AGausmann, 2016-11-10:
Yep, our team has one. I'll get started then

rhauch, 2017-01-03:
I have a [pull request](https://github.com/strongback/strongback-java/pull/88) (targeted for the Strongback 2017.1.0 version) that isn't quite ready that will add 3rd party libraries for CTRE TalonSRX library and the navX-MXP library. The latter version that is compatible with the 2017 WPILib for Java will be officially released on 1/6/2016, so the initial version of Strongback 2017.1.0 will add both libraries and the new Strongback CLI will properly install them into the WPILib's user library directory. This work should build upon that work.

rhauch, 2017-01-07:
The `master` branch now includes the navX-MXP library (see #88), the build already includes it in the artifacts, and the Strongback CLI will automatically install the library as a WPILib _user library_. All that remains is to add the APIs to Strongback to make use of the navX-MXP library.